CRLG/LABOTBOX #14: [ExternalControler] Donner la capacité à Labotbox d'être piloté depuis une application externe

NicoFarmer, 2018-07-25:
Créer un basic module ExternalControler permettant à une application extérieure (client python par exemple) de piloter Labotbox.
Parmi les pilotages possibles il faut au minimum pouvoir : 
    - Modifier la valeur d'une variable du DataManager
    - Lire  la valeur d'une variable du DataManager

Le nouveau module implémentera un serveur TCP, permettant de piloter Labotbox depuis un PC distant.

Créer une classe python LabotboxClient mettant en oeuvre une API simplifiée pour accéder au serveur.
Un nouveau dépôt git accueillera cette classe python ainsi que des exemples de prise de contrôle à distance de Labotbox.

laguiche, 2018-07-26:
On pourrait utiliser la même classe que celle de communication rs232 et la dériver selon le type de connexion choisie. Quel type de socket comptes tu prendre pour le serveur ?

NicoFarmer, 2018-07-26:
L'idée principale de ce nouveau module, c'est de pouvoir scripter en local ou depuis un PC distant.

C'est vrai qu'avoir un serveur générique avec le choix de l'interface (RS232, UDP, TCP, CAN, I2C, ...) est toujours bien dans le principe.
Dans la pratique, je ne suis certain qu'on change d'interface aussi souvent que ça pour scripter.
Pour cette raison, je souhaite plutôt privilégier une interface donnée mais surtout fournir des exemples de prise de contrôle en python par exemple.
L'architecture du module sera construite pour faire évoluer l'interface de communication sans tout casser en cas de besoin.

Comme je l'ai indiqué dans le ticket, j'ai choisi TCP qui me paraissait le plus naturel pour la mise en place de requêtes/réponses entre un client et un serveur.
Le choix d'un serveur TCP mettra naturellement en oeuvre  des sockets de type TCP.

laguiche, 2018-07-26:
Une websocket serait envisageable ?

NicoFarmer, 2018-07-31:
A vu de pif, ça me parait carrément faisable, les websockets étant fortement liées à du TCP.
L'API Qt pour mettre en oeuvre les websocket ressemble fortement à celle du TcpServer que j'utilise déjà pour le serveur.

Tu verras l'architecture que j'ai mis en place pour cette partie permet de faire évoluer facilement le type de connexion.

En gros, je confirmerai ça par quelques essais, mais il ne semble pas y avoir de soucis.

NicoFarmer, 2018-08-04:
Voici une première version du nouveau Module **ExternalControler**
Ce module permet à certaines fonctions de Labotbox d'être pilotées depuis une application externe.
2 types de connexion sont disponibles pour le commit de ce jour : 
    **- Tcp
    - WebSocket**
Le UserGuide du module explique les paramètres de configuration du fichier EEPROM.ini à positionner pour configurer une connexion Tcp ou WebSocket. 
Il décrit aussi l'API des échanges entre Labotbox et le monde extérieur.

Un script Python est donné comme exemple dans le répertoire /BasicModules/ExternalControler/client_example : **example_labotbox.py**
Le choix du type de connexion (Tcp, WebSocket), de l'adresse IP et du numéro de port se font au début du fichier. Le script est compatible avec les 2 types de connexions.

Un fichier **example_websocket_labotbox_page.html** est fourni pour tester le contrôle par WebSocket depuis un navigateur web.
Ouvrir le fichier .html avec un navigateur, établir la connexion avec Labotbox et envoyer les instructions.

closed 2018-08-04 by NicoFarmer