CRLG/LABOTBOX #16: Certaines fonctions plantent lorsque le projet est linké avec opencv sous Robuntu

NicoFarmer, 2018-07-26:
Lorsqu'on ajout le modume BotCam, il est nécessaire de linker le projet avec opencv.
Dans ce cas, certaines fonctionalités de Labotbox plantent l'application : 
    - Le changement de couleur de fond d'une fenêtre par exemple

Condition : 
  Robuntu 16.04
  Qt 5.8
  Opencv 2.4.9.1
 
Le problème apparait aussi bien si la compilation est faite en debug qu'en release
Le message d'erreur remonté en console de debug pendant l'exécution  est :
___________________________________________________________________
_(LaBotBox:8996): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(LaBotBox:8996): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(LaBotBox:8996): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(LaBotBox:8996): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(LaBotBox:8996): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(LaBotBox:8996): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(LaBotBox:8996): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(LaBotBox:8996): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(LaBotBox:8996): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(LaBotBox:8996): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'_
___________________________________________________________________

laguiche, 2018-07-26:
Quels sont les paramètres du fichier de configuration (.ini) pour le module de caméra ?

NicoFarmer, 2018-07-26:
Ce sont les paramètres par défaut du module (que je n'ai pas modifié).
Je ne pense pas que ça vienne du module BotCam en lui même mais plutôt un effet de bord un peu exotique du fait de linker opencv avec les libs Qt (j'ai fouillé vite fait sur le net, ça n'a pas l'air de se résoudre en 2 secondes).

Voici la section du fichier .ini liée à BotCam : 
___________________________________________________________
[BotCam]
background_color=@Variant(\0\0\0\x43\x1\xff\xff\xf0\xf0\xf0\xf0\xf0\xf0\0\0)
geometry=@Rect(463 93 504 562)
listeCouleurs=rouge
listeEcartCouleur=100
listePurete=100
listeSaturation=100
listeSmax=100
listeSmin=100
niveau_trace=4294967295
visible=true
___________________________________________________________

NicoFarmer, 2018-07-31:
Complémént d'info : 
Dans le module UserGuides, il y a une fonctionnalité qui permet d'éditer une page html pour écrire le guide d'utilisation d'un module. Dans cette fenêtre, il y a possibilité de changer la couleur du texte. 
Dès qu'on appui sur le bouton pour faire apparaître de sélection de couleur, l'application plante.
Ca rejoint le descriptif principal ou déjà la fenêtre de sélection de couleur était source de problème.

NicoFarmer, 2018-08-21:
Le problème n'apparait plus dans lernière configuration en date : 
Robuntu18.04.1
Qt 5.11.1
OpenCV 3.2

Dès que cette configuration sera déployée auprès des membres du club, le ticket pourra être soldé.

NicoFarmer, 2018-09-04:
Pas de problème constaté avec la configuration **Raspberry Pi 3** :
RobianPi v0.1
Qt 5.7
OpenCV 3.2

NicoFarmer, 2018-10-31:
Le problème n'apparait plus dans les 2 configurations déployées au club depuis plusieurs semaines : 

- Sur PC avec Robuntu18.04
- Sur RaspberryPi3 avec RobianPi

Le ticket est soldé.

closed 2018-10-31 by NicoFarmer